raulmur/ORB_SLAM2 #872: [Doc] Explanation of principles to select key frames desired

yiakwy, 2020-03-05:
System environment: Monocular Camera with structures estimated from motion

The system says will keep features extracted from "key frames". Despite "initial frame" and "the current frame", how we choose key frames from.

It seems that frames in between is only used for camera motion and pose estimation (see implementation in "MonocularInitialization" module).

Hence, is there any principles to keep frames?

AlejandroSilvestri, 2020-03-05:
@yiakwy 

You are right.  Frames are mainly used to track position and then discarded, only a few of them are upgraded to Keyframe and kept in map.

[Tracking::NeedNewKeyFrame()](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Tracking.cc#L977) is the one responsible to decide when to make a keyframe.  In the code you can see the decision process.

In [these lines](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Tracking.cc#L1028) you can read the criteria.  Here's a copy:

    // Condition 1a: More than "MaxFrames" have passed from last keyframe insertion
    // Condition 1b: More than "MinFrames" have passed and Local Mapping is idle
    // Condition 1c: tracking is weak
    // Condition 2: Few tracked points compared to reference keyframe. Lots of visual odometry compared to map matches.

    if((c1a||c1b||c1c)&&c2) lets insert a new keyframe

1. 

1. 

1.

yiakwy, 2020-03-06:
@AlejandroSilvestri  Thank you! It helps me.