raulmur/ORB_SLAM2 #878: ORB_SLAM2 ROS Package in virtual machine VMware (Ubantu 16.04 Kinet ROS)

dilipdhapse, 2020-03-23:
I have complied ORBSLAM2 package from
 https://github.com/raulmur/ORB_SLAM2

and followed by all installation given in guidelines. 

I have installed Pangoline, Eigen 3.1.0 and DBoW and g2g given in ORB_SLAM2 Package.

all instillation was successful when I ran:
$ rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/KITTI00-02.yaml false

I am getting one warning msg. 

''Framebuffer with requested attributes not available. Using available framebuffer. You may see visual artifacts.X11 Error: GLXBadFBConfig ''

after that I have started rosnode to publish stero images on two topic :
/camera/left and right /image 

ORB_SLAM node is getting crashed showing msg 'core segmentation falt'

this terminal is getting closed.

what need to be done is it because of virtual machine.

I have done same installation on NVDIA Jetson Tx2 all works fine but it is not working on Vmware (Ubantu Kinet ROS)machine installed on W10 machine.

Will you please guide me in this issue.

Thanks,
Dilip Dhapse

xujinjinjinjin, 2020-12-18:
Did you solve this problem? Will you please guide me in this issue?Thank you.

xujinjinjinjin, 2020-12-20:
> Did you solve this problem? Will you please guide me in this issue?Thank you.

I bypassed it by going in the g2o cmakelists and the orb slam cmakelists and removing -march=native .yeah~