raulmur/ORB_SLAM2 #887: ORB_SLAM2 build error - OpenCV not found

DimitarVel, 2020-04-16:
I have built all the required dependencies listed in the Dependencies.md file including OpenCV v 4.3.0 (latest). When i tried to build ORB_SLAM2 in a /build/ folder using the following command:
`sudo cmake -S ../ -B ./`
I got the following error:
```
CMake Error at CMakeLists.txt:35 (message):
  OpenCV > 2.4.3 not found.
```

I googled and found an existing [issue](https://github.com/raulmur/ORB_SLAM2/issues/527#issue-300055227) which basically set up the lib paths in env, so i did that but it didnt work out for me. 

OS: Linux (Manjaro)
OpenCV: 4.3.0 (without contrib)
CMake: 3.16.4
All other dependencies have the latest version.

ahmed-andalib, 2020-05-03:
I had issues with OpenCV 4.3 so I followed this to resolve it:

https://cv-tricks.com/installation/opencv-4-1-ubuntu18-04/

After that I modified the CMakeLists.txt under ThirdParty/DBow2 to check for version 4.3, this worked.
However I am having issues with Eigen, I tried to download an compile it followed byt sudo apt-gt install to install Eigen from official repo. But keep getting this error:

Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.3.7") 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
EIGEN3_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/asa/orb_slam/ORB_SLAM2/Thirdparty/g2o
   used as include directory in directory /home/asa/orb_slam/ORB_SLAM2/Thirdparty/g2o
   used as include directory in directory /home/asa/orb_slam/ORB_SLAM2/Thirdparty/g2o

In the final section of build.sh, you may get some OpenCV include errors like: 
#include <opencv/cv.h>
I haven't fixed them but I think they can be fixed by changing appropriate includes, for my machine
#include <opencv2/core.hpp> works on other programs

Please let me know if you were able to fix the EIGEN_INCLUDE_DIT problem

thanks

ahmed-andalib, 2020-05-04:
Update I was able to fix both Eigen3 and OpenCV issues.

OpenCV: Downloaded and installed OpenCV 3.2, updated the pkgconfgig opencv.pc file as mentioned in the above link I gave before to 3.2 one
Eigen3: just installed Eigen from Ubuntu official repo. 

afterwards compilation raised a couple of errors: the fixes are simple just follow this:
http://blog.leanote.com/post/gaunthan/ORB-SLAM2-compiling-problems

after fixing the compiler errors I was able to compile the source

whoakashpatel, 2025-03-14:
@ahmed-andalib Can you provide me the flow of things you've done to fix those issues?  I have installed eigen3.4.0, and have open4.5.0 which I'm trying to downgrade to 3.2 but it has c vs cpp conflicts. I'm trying to work with DynaSLAM but hardstuck in building it. It's based on ORBSLAM2 itself. 
I was first getting all those EIGEN include and then DEPRECATION errors, which I somehow fixed, but now there was some opencv conflict cuz opencv2 is installed at /usr/local/include/opencv4/opencv2 and this causes
I'm quite new to these things. A document would help.