raulmur/ORB_SLAM2 #897: Covisibility Graph - updating connections

dybas133, 2020-05-21:
Acording to the paper (page 4 section D) the Covisibility Graph's connections are supposed to be both ways and get created if and only if there are at least 15 shared map points.

However by looking at the code at [KeyFrame::UpdateConnections()](https://github.com/raulmur/ORB_SLAM2/blob/master/src/KeyFrame.cc#L318#L367) I've noticed that  connections from others to KeyFrame being updated are added if there are at least 15 common map points, as supposed, but KeyFrame has this `mConnectedKeyFrameWeights = KFcounter` at line 367, where all KeyFrames that have at least one common map point are added.

Is this intended or is it a bug?

The way it is now it is not undirected graph and does not keep those 15 and above common map points.

zeal-up, 2023-12-07:
@dybas133 Hi, I have also noticed this "bug?". Have you get more knowledge about this?

dybas133, 2023-12-11:
Hi @zeal-up,

Wow, that was such a long time ago.
For over a year I haven't been working with orb slam anymore, but I remember fixing it by myself and it seemed to improve the algorithm's memory management.
Also the resulted maps had less weird point clumps.

I don't have access to the code anymore, so I can't provide concrete result, but I'll try to bring it up, because I remember the way of thinking I had (give me a few days, I'm quite busy at the moment).