raulmur/ORB_SLAM2 #903: Failed to load image

gizemozd, 2020-05-31:
NOTE: Issue is the same as the previous ones, but the solution was unrelated to mine. 

OS: Ubuntu 14.04 
OpenCV: 2.4.8
All dependencies installed successfully and I could run Euroc examples without any problems.

Now I am trying to try it with a different dataset. I saved the data in a directory called "Dataset" in the ORB_SLAM2 project root directory. mono_cold is the same as mono_euroc. I created the timestamps.txt and data.csv file accordingly and run it as follows:

**./Examples/Monocular/mono_cold Vocabulary/ORBvoc.txt Examples/Monocular/cold.yaml ~/ORB_SLAM2/Database/mav0/data Examples/Monocular/EuRoC_TimeStamps/cold.txt**

And I keep getting this error:

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 3.637
- fy: 3.638
- cx: 320
- cy: 240
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 1598

**.pnged to load image at: /home/gizem/ORB_SLAM2/Database/mav0/data/1403636579763555584**

Weird thing is that it says ".pnged to load" instead of "failed to load". Names in data.csv and data file are compatible with each other and modified according to the Euroc dataset. I really appreciate it if anyone would help!

gizemozd, 2020-05-31:
I also got this error before.

[image]

stolpa4, 2020-06-01:
> NOTE: Issue is the same as the previous ones, but the solution was unrelated to mine.
>
> OS: Ubuntu 14.04
> OpenCV: 2.4.8
> All dependencies installed successfully and I could run Euroc examples without [...]

Did you try to compile some test code with a plain opencv read image function, which takes, for example path to the dataset and just reads an image from it?
.pnged to load seems to be a multithreading issue, when another thread in parallel puts something like carriage return symbol to the console, while this thread still prints error

esteimle, 2020-06-19:
I had this same problem with a bunch of files I had from windows.  For some reason there was a trailing invisible \r in all my filenames.  I somewhat lazily solved it by modifying the loadimages function in the stereo_euroc example:

`void LoadImages(const string &strPathLeft, const string &strPathRight, const string &strPathTimes,
                vector<string> &vstrImageLeft, vector<string> &vstrImageRight, vector<double> &vTimeStamps)
{
    ifstream fTimes;
    fTimes.open(strPathTimes.c_str());
    vTimeStamps.reserve(5000);
    vstrImageLeft.reserve(5000);
    vstrImageRight.reserve(5000);
    while(!fTimes.eof())
    {
        string s;
	string temp;
        getline(fTimes,s);
        if(!s.empty())
        {
            stringstream ss;
            ss << s;
	    temp = strPathLeft + "/" + ss.str();
	    temp = temp + ".png";
	    temp.erase( std::remove(temp.begin(), temp.end(), '\r'), temp.end() );
//	    cout << "temp: " << temp << endl;
	    //vstrImageLeft.push_back(strPathLeft + "/" + ss.str() + ".png");
            vstrImageLeft.push_back(temp);
            temp = strPathRight + "/" + ss.str();
            temp = temp + ".png";
            temp.erase( std::remove(temp.begin(), temp.end(), '\r'), temp.end() );
//	    vstrImageRight.push_back(strPathRight + "/" + ss.str() + ".png");
	    vstrImageRight.push_back(temp);
            double t;
            ss >> t;
            vTimeStamps.push_back(t/1e9);

        }
    }`

gizemozd, 2020-06-19:
@esteimle Thanks for your answer, it's a smart solution though! 
I also solved this issue but forgot to update it here. Similar to your case, there was an invisible character in my timestamp file, I realized it when I opened it up on NotePad. I erased them now it works fine.

Oragold, 2023-08-15:
@esteimle Every time I execute a command, the 'png' that should have been followed by the image name disappears, but appears at the 'fai' on the next line. I copied the MH01.txt file from Ubuntu to WIN's TXT and found many extra spaces.
The switch behavior under Ubuntu is  \n, while under WIN is  \r\n, which may be the reason for this problem. Thank you for helping me solve it with your code.

Oragold, 2023-08-15:
@esteimle Well，terminal is ok, but still can't read photos. Maybe because of opencv 's imread func.