raulmur/ORB_SLAM2 #909: Creating Vocabulary as Txt file

manthan99, 2020-06-19:
I want to create my own vocabulary txt file. Is there any demo code similar to that provided in the original [DBoW2 Repo](https://github.com/dorian3d/DBoW2/tree/master/demo)? The original DBoW2 Repo only has the option to save as yml file. 
As pointed out in #9 to use the saveToTextFile Function, it is not available in [DBoW2 Repo](https://github.com/dorian3d/DBoW2) and there is no available demo file in the [ORB_SLAM2 Repo](https://github.com/raulmur/ORB_SLAM2/tree/master/Thirdparty/DBoW2) 

 #9 #724

stolpa4, 2020-06-26:
@manthan99 You can just create your vocabulary as usual with the original DBoW2 repo, save it as yaml, for example. And then use the function save as text file from Thirdparty/DBoW2 of this project to convert it from yaml to txt. Totally accepted option.

manthan99, 2020-06-26:
@stolpa4 Thankyou for your reply. I will try it out

manthan99, 2020-06-26:
@stolpa4 I am unable to compile my code which uses the function save as text file. I have created the vocabulary using the demo.cpp file and saved it as "castle_ruins.yml.gz"

Here are the lines which I am using to convert. Are these correct? 

` TemplatedVocabulary<DBoW2::FORB::TDescriptor, DBoW2::FORB> voc("castle_ruins.yml.gz"); //to load the vocabulary 
  voc.saveToTextFile("castle_ruins.txt"); //save as text file`

I have included the following in the code-

`#include "TemplatedVocabulary.h"`
`#include "BowVector.h"`
`#include "FeatureVector.h"`
`#include "FORB.h"`

Yet I am getting the following error-
`
 error: ‘FORB’ is not a member of ‘DBoW2’
   TemplatedVocabulary<DBoW2::FORB::TDescriptor, DBoW2::FORB> voc("castle_ruins.yml.gz");
                              error: ‘FORB’ is not a member of ‘DBoW2’
`

Could you provide me with the working code if you have one.

stolpa4, 2020-06-26:
@manthan99 
Use Thirdparty/DBoW2 from this repo for the conversion.
1) Build the Thirdparty/DBoW2 library itself
2) Link it against it like this: 
```
target_link_libraries(${PROJECT_NAME}
${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
)
```
3) Include [this file](https://github.com/raulmur/ORB_SLAM2/blob/master/include/ORBVocabulary.h) only.
4) Use `ORBVocabulary` instead of `TemplatedVocabulary<DBoW2::FORB::TDescriptor, DBoW2::FORB>`

Try this

manthan99, 2020-06-26:
Hii @stolpa4 
Thank you very much for your help. I am able to save the vocabulary as a txt file using this -
> @manthan99
> Use Thirdparty/DBoW2 from this repo for the conversion.
> 
> 1. Build the Thirdparty/DBoW2 library itself
> 2. Link it against it like this:
> 
> ```
> target_link_libraries(${PROJECT_NAME}
> ${PROJECT_SOURCE_DIR}/Thirdparty/DBoW2/lib/libDBoW2.so
> )
> ```
> 
> 1. Include [this file](https://github.com/raulmur/ORB_SLAM2/blob/master/include/ORBVocabulary.h) only.
> 2. Use `ORBVocabulary` instead of `TemplatedVocabulary<DBoW2::FORB::TDescriptor, DBoW2::FORB>`
> 
> Try this

Only left to check how well is this new dictionary working in my environment. 
Just to clarify, we need to keep the branching factor(K) = 10 and depth(L) = 6, right?

manthan99, 2020-07-16:
Have a look at this repo for step by step procedure to create your own vocabulary:
https://github.com/manthan99/ORB_SLAM_vocab-build

closed 2020-07-16 by manthan99

XinnWang, 2021-10-12:
@stolpa4 i want to use a smaller factor and depth, which part of the source should i change accordingly ?