raulmur/ORB_SLAM2 #911: Raspberry 4 speed and ComputeKeyPointsOctTree(allKeypoints)

Fredrum, 2020-06-24:
Hi!
I'm running this on a RPi4 and am measuring how long the different stages take to execute because I'm hoping to be able to make it run at 30fps or even faster.

I notice that the function ComputeKeyPointsOctTree(allKeypoints) inside ORBExtractor takes ~60ms.
And the loop a bit further down that steps through the image pyramid and does computeDescriptors() take around 45ms.

Now I also notice that there's a commented out 'ComputeKeyPointsOld()' that I haven't tried to re-enable.

My question is, what does the expensive ComputeKeyPointsOctTree() function do and is it really necessary!?  :)

Are there any shortcuts that I could consider here, while maybe accepting some loss of quality to the result in the pursuit of getting higher fps?

Cheers
Fred

stolpa4, 2020-06-24:
@Fredrum it is totally necessary for the slam to match features robustly. It performs keypoint culling and distribute all the useful keypoints uniformly among all the image.
You can try the old function.
Also you can go with a simple opencv orb extractor, but the quality of tracking will decrease dramatically

Fredrum, 2020-06-24:
@stolpa4  Ok well I wouldn't want drastic decrease in tracking quality so will have to try and think of other ways to optimize.
For example I'll try to replace the fast() calls with this asm version,    https://github.com/flexbex/fast9_arm_asm
Don't know if it can replace the currently used fast() but we'll see. It's quicker for me to just test it rather than to try and fully understand the code first.

Also do you think that the OctTree algorithm can be improved with something like FLANN?  (I think I saw this mentioned as a suggestion)  https://github.com/mariusmuja/flann    or     https://summerofcode.withgoogle.com/archive/2017/projects/6710140954738688/

stolpa4, 2020-06-25:
@Fredrum 
1) How many cores / hardware threads does your device have?
There are 3 basic threads in the ORB-SLAM2: Local Mapping, Loop Closer + Main thread (tracking). For visualization a separate 4th thread is used. Maybe some of them are of no use for you? For example, I doubt the usefulness of Loop Closure at all for automotive industry, as a usual car trajectory is not looped. And maybe you're good with some simple plotting software or a program patch to plot the trajectory itself without map points and stuff like that (it can plot one point at each frame - super cheap, comparing with Pangolin). 
2) OctTree algorithm does use quadTree. Just enough features is anchored to nodes with this function. (it starts with 3 base nodes and then spawn them, every node has ROI and if features are exactly in this ROI, they classified as belonging to this node). The algorithm leaves the most responsive features only, all the rest are flushed. As you see, there are a lot of room for optimization, you can think in the following direction: how not to detect useless features (that will be discarded by an OctTree function later on). Or a similar stuff.
3) Pay attention [to this line](https://github.com/raulmur/ORB_SLAM2/blob/master/src/ORBextractor.cc#L769)
It defines a parameter that is responsible for grid image division (all these algorithms do work with a patch of image, not the whole image). You can adjust this parameter (increasing = less grid patches), but make sure you are changing it in some reasonable limits (to be defined by you for your particular case). It can improve the performance, with not so drastic tracking quality drop, but I had some problems with the global optimization when adjusted it some time ago (maybe minor thing, not sure).
4) The code quality itself is not good enough, also there are lots of memory leaks (easily discovered by valgrind). I personally think the assembler may make things even worse.

Fredrum, 2020-06-25:
@stolpa4 

Thank you so very much for all that advice and insightful suggestions!  :D

I'll will have a go and try to do some of the things that you suggest.
This is an 'after work' kind of project for me that I'm slowly chipping away at.

The Raspberry Pi 4 that I'm currently using has four CPU cores and a separate not too strong GPU.
It seems there's no CUDA for the Rapberries but some people use the NEON architecture to optimise some things and I have found some libraries of lower level Math type operations written for NEON.  But that might be more work than I'm wanting to do and too hardware specific. I'd rather do broader things that would be more generally applicable.

I don't really need the visual part for my intended end use. Not sure if it impacts the fps that much but I'll try disabling and see.
I understand that it's the ORBExtraction+Tracking thread that is the main thing to optimise first as it's constantly 'ticking', while the other two threads just do shorter bursts of work. Though I suppose I'd want to smooth out those bursts also, to avoid fps spikes.

Cheers

Janebek, 2022-07-18:
Hi ！ How is your work now ？ @Fredrum