raulmur/ORB_SLAM2 #912: Trouble understanding CameraTrajectory.txt for 3D reconstruction

esteimle, 2020-06-25:
I have orbslam stereo working well with my own calibrated stereo camera pair.  I used it to walk around an object I want to scan and the path lines up with me getting up from my desk, walking over to the object, doing a 360 around it and walking back.  I'm also able to generate point clouds from each pair of stereo images using openCV.  But now I was hoping to use the cameraTrajectory data to help line up my point clouds with each other but I keep failing.  I understand that the data (based on euroc sample application) is in the form: timestamp x,y,z of camera qx, qy, qz, qw of the camera.

I was using PCL, they have a pcl::transformPointCloud that takes the cloud, translation vector and quaternion rotation.  I picked two frames about 100ms apart, transform the first point cloud by it's matching vector and quaternion,  and then transform the second by its matching vector and quaternion. That didn't work so I've tried adding the vectors and multiplying the rotation up to that point.  This resulted in a what looked like a hurricane eye.  I tried subtracting the two vectors and multiplying the first quaternion by the inverse of the second.  I've tried negating the second vector and inverting the second quaternion.  But no matter what I try the two just never line up.  I don't expect it to be perfect but most of my attempts put the two clouds further away than if I just combined them with no rotation.

Any idea what I might be missing here?

Thank you.

Fredrum, 2020-06-25:
Not a straight answer here and sorry if this is too obvious,

https://github.com/raulmur/ORB_SLAM2/issues/766

https://github.com/raulmur/ORB_SLAM2/issues/195

esteimle, 2020-06-26:
Thank you.  I've actually come pretty close just using the quaternions and the translation, but it still isn't exact.  I was wondering if I understood the trajectory file correctly and if the approach of rotating and translating each point cloud by the values for each ones timestamp was the correct approach.

zyuanbing, 2021-03-07:
I had the exact same problem as you described. Have you solved this problem?