raulmur/ORB_SLAM2 #913: NULL pointers in the array returned by GetTrackedMapPoints()

isVoid, 2020-06-25:
Hi, I'm very new to ORB_SLAM2. If I understand it correctly, `GetTrackedMapPoints()` returns an array that contains map points tracked from the **current frame**. When I just booted the system, I tested the size of the return array, which is non-zero, but the array contains null pointers. Why is this happening?

My sensors are a set of stereo camera. I called the method in `Node::Update()`, in the line after `PublishMapPoints()`, my goal is to publish the map points of current frame as a topic, in the same manner as how `GetAllMapPoints()` is handled.

isVoid, 2020-06-26:
I guess this behavior is expected. The array is allocated with all possible map points, and the outliers were dropped (set to NULL).

Ali-fe, 2022-05-10:
what did you do for this problem?

Ali-fe, 2022-05-10:
I have like this problem in SLAM3