raulmur/ORB_SLAM2 #915: KITTI Mono ROS Example works but KITTI Stereo ROS Example doesn't work

mhaboali, 2020-07-01:
Hi and thanks a lot for this great code.

When I used a rosbag contains left and right gray images topics and mapped topics names to the expected ones from the Stereo node. When I run this command `rosrun ORB_SLAM2 Stereo $PWD/Vocabulary/ORBvoc.txt $PWD/Examples/Stereo/KITTI03.yaml false`

SLAM doesn't work at all, it just shows me the images without any features and without odometry data(camera poses) as shown below:

[image]

And as you can see I always get this log in the terminal:
```
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 0 points
Track lost soon after initialisation, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New map created with 0 points
```

Could anyone help with this case? I've looked a lot for a solution but it seems there's no similar case except for this issue https://github.com/raulmur/ORB_SLAM2/issues/828

Turing47, 2021-11-09:
Hi, Did you check the name of the nodes subscribed:
    message_filters::Subscriber<sensor_msgs::Image> left_sub(nh, "/camera/left/image_raw", 1);
    message_filters::Subscriber<sensor_msgs::Image> right_sub(nh, "/camera/right/image_raw", 1);
I guess you may make mistake here

withnoerr, 2024-07-12:
Did you solve that?