raulmur/ORB_SLAM2 #916: Makefile:83: recipe for target 'all' failed make: *** [all] Error 2

EasternH, 2020-07-02:
CMakeFiles/ORB_SLAM2.dir/build.make:86: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/ORB_SLAM2.dir/build.make:494: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:62: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/System.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:134: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:110: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o] Error 1
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I wonder where wrong about my installation......,what need I do.

FrancescoFallica, 2020-07-07:
These error messages are not enough, you need to be more specific for understanding the problem and trying to help you.

EasternH, 2020-07-10:
> These error messages are not enough, you need to be more specific for understanding the problem and trying to help you.

OK , I have found the solution.

srujanpanuganti, 2020-07-29:
I am encountering the same error. How did you solve it? @EasternH

blarkj, 2020-08-05:
I am having the same error. Can you please elaborate @EasternH ?

rfzeg, 2020-08-11:
> > These error messages are not enough, you need to be more specific for understanding the problem and trying to help you.
> 
> OK , I have found the solution.

Could you bother to share how you fixed it?