raulmur/ORB_SLAM2 #919: ORB-SLAM detects new keypoints with delay

zanazakaryaie, 2020-07-11:
Hi. I'm using ORB-SLAM on a live USB camera. It works fine but if I move the camera just a little bit fast to explore new regions, it loses the track. I noticed that after initialization If I move the camera a little bit fast, it can't add new keypoint in new frames unless I move the camera very slowly and hold it in the new regions waiting for the ORB-SLAM to detect new keypoints. 
How can I solve this issue? Let me mention that I have decreased the fast threshold to 5, and increased the maximum number of ORB features to 5000. Also, I lowered Viewer.KeyFrameSize to 0.05. I guess it results in faster initialization.
Any help is highly appreciated

zanazakaryaie, 2020-07-11:
It seems that ORB-SLAM only tracks new points if the current frame has enough similarity to the reference keyframe. This similarity is measured by BoW. So, I think when the camera is moved fast, the new frame will be very dissimilar to the reference keyframe and no PnP is set up. Anybody there?

xvzezi, 2020-07-16:
Lots of reasons.
But for fast motion, motion blurs will lead to bad keypoints match, which could lead to lost.
Besides, ORB-SLAM2 provides several tracking model, which can be found in class Tracking

zanazakaryaie, 2020-07-18:
Thanks @xvzezi 
The motion is not that fast to cause motion blur. I have seen Tracking.cc and it seems that ORB-SLAM requires enough translation between frames to add new points. That's why when I rotate the camera even with slow motion, it loses the track because it can't add new map points.

xvzezi, 2020-07-22:
I notice that you said ***Rotate the camera***
for monocular initialization, translations needed to provide enough parallax. Pure rotation or when rotation dominate the motion, it will fail. 
because you cannot triangulate points for monocular camera with pure rotation.

pengyuan-zhou, 2020-08-17:
> I notice that you said _**Rotate the camera**_
> for monocular initialization, translations needed to provide enough parallax. Pure rotation or when rotation dominate the motion, it will fail.
> because you cannot triangulate points for monocular camera with pure rotation.

Hi @xvzezi , nice point, does Stereo and RGB-D also face same problems with that?

xvzezi, 2020-10-21:
> > I notice that you said _**Rotate the camera**_
> > for monocular initialization, translations needed to provide enough parallax. Pure rotation or when rotation dominate the motion, it will fail.
> > because you cannot triangulate points for monocular camera with pure rotation.
> 
> Hi @xvzezi , nice point, do Stereo and RGB-D also face same problems with that?

Nope. Triangulation is used for computing depth initialization.

alexistex, 2024-01-01:
only use orb_slam3 -> cpp code and very fast  , without delay