raulmur/ORB_SLAM2 #922: Failed to build orbslam2 in ubuntu 18.04

shumshersubashgautam, 2020-07-15:
I stuck on this error any idea on this error:

/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to _glapi_tls_Current' collect2: error: ld returned 1 exit status CMakeFiles/mono_euroc.dir/build.make:165: recipe for target '../Examples/Monocular/mono_euroc' failed make[2]: *** [../Examples/Monocular/mono_euroc] Error 1 CMakeFiles/Makefile2:294: recipe for target 'CMakeFiles/mono_euroc.dir/all' failed make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2 make[1]: *** Waiting for unfinished jobs.... /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to _glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_tum.dir/build.make:165: recipe for target '../Examples/Monocular/mono_tum' failed
make[2]: *** [../Examples/Monocular/mono_tum] Error 1
CMakeFiles/Makefile2:220: recipe for target 'CMakeFiles/mono_tum.dir/all' failed
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to _glapi_tls_Current' collect2: error: ld returned 1 exit status CMakeFiles/rgbd_tum.dir/build.make:165: recipe for target '../Examples/RGB-D/rgbd_tum' failed make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1 CMakeFiles/Makefile2:109: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2 /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to _glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/mono_kitti.dir/build.make:165: recipe for target '../Examples/Monocular/mono_kitti' failed
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
CMakeFiles/Makefile2:257: recipe for target 'CMakeFiles/mono_kitti.dir/all' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
/usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to _glapi_tls_Current' collect2: error: ld returned 1 exit status CMakeFiles/stereo_kitti.dir/build.make:165: recipe for target '../Examples/Stereo/stereo_kitti' failed make[2]: *** [../Examples/Stereo/stereo_kitti] Error 1 CMakeFiles/Makefile2:146: recipe for target 'CMakeFiles/stereo_kitti.dir/all' failed make[1]: *** [CMakeFiles/stereo_kitti.dir/all] Error 2 /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libOpenGL.so: undefined reference to _glapi_tls_Current'
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_euroc.dir/build.make:165: recipe for target '../Examples/Stereo/stereo_euroc' failed
make[2]: *** [../Examples/Stereo/stereo_euroc] Error 1
CMakeFiles/Makefile2:183: recipe for target 'CMakeFiles/stereo_euroc.dir/all' failed
make[1]: *** [CMakeFiles/stereo_euroc.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

xvzezi, 2020-07-16:
run
./build.sh > build_log.txt 2>&1
to get full error output in 'build_log.txt'

esteban-andrade, 2020-08-08:
Hi I am facing the same issue. I am attaching a log file
[build_log.txt](https://github.com/raulmur/ORB_SLAM2/files/5045525/build_log.txt)

xvzezi, 2020-08-12:
> Hi I am facing the same issue. I am attaching a log file
> [build_log.txt](https://github.com/raulmur/ORB_SLAM2/files/5045525/build_log.txt)

usleep cannot be found, 
this is a solved problem, you should search for other similar issues first.
just include a header file unistd.h on linux

esteban-andrade, 2020-08-12:
> > Hi I am facing the same issue. I am attaching a log file
> > [build_log.txt](https://github.com/raulmur/ORB_SLAM2/files/5045525/build_log.txt)
> 
> usleep cannot be found,
> this is a solved problem, you should search for other similar issues first.
> just include a header file unistd.h on linux

Hi That Seems to fix the issue. Also for reference, the ROS part didn't build. The solutions were to add boost libraries and packages for the Cmake Lists for the ROS option. 

Cheers,

linpeisensh, 2020-12-12:
@xvzezi 
hello, I am facing the same issue.
Here is my log file.
[build_log.txt](https://github.com/raulmur/ORB_SLAM2/files/5683114/build_log.txt)
I have edited as this link and it doesn't work:
https://github.com/raulmur/ORB_SLAM2/issues/535#issuecomment-372008559
Could you please help me?
Thanks!