raulmur/ORB_SLAM2 #944: ORB_SLAM2 ROS Segmentation fault (core dumped) on Ubuntu 16.04 running on VM

mohGhazala96, 2020-09-28:
I am using Ubuntu 16.04 on parallels Virtual machine running on MacOS 10.15.5. I have built both the build.sh and the build_ros.sh successfully. 
I have OpenCV 2.4.3 installed and Eigen3 3.1.0. When I run the Monocular Examples using the TUM Dataset, it works just fine. 
My problem is when running the Mono AR ROS example:
`parallels@ubuntu:/media/psf/Work/ORB_SLAM2$ rosrun ORB_SLAM2 MonoAR Vocabulary/ORBvoc.txt Examples/ROS/ORB_SLAM2/Asus.yaml 
Segmentation fault (core dumped)`
I tried changing the camera parameters in the Asus.yaml file but it's not working.

mohGhazala96, 2020-09-28:
> I am using Ubuntu 16.04 on parallels Virtual machine running on MacOS 10.15.5. I have built both the build.sh and the build_ros.sh successfully.
> I have OpenCV 2.4.3 installed and Eigen3 3.1.0. When [...]
Segmentation fault (core dumped)`
> I tried changing the camera parameters in the Asus.yaml file but it's not working.

makolele12, 2021-08-19:
Hi @mohGhazala96 , I just met with the same problem while trying to replay the EuRoc V101 rosbag, did you find any solution?

northpoleforce, 2024-03-25:
@mohGhazala96 @makolele12 
I met the same problem
I fixed it by the solution below
change `ORB_SLAM2/Examples/ROS/ORB_SLAM2/CMakeLists.txt`
from
```
set(LIBS 
${OpenCV_LIBS} 
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM2.so
)
```
to
```
set(LIBS 
${OpenCV_LIBS} 
${EIGEN3_LIBS}
${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM2.so
/usr/local/lib/libopencv_core.so.3.2
/usr/local/lib/libopencv_imgproc.so.3.2
/usr/lib/x86_64-linux-gnu/libopencv_core.so.4.2
/usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2
-lboost_system
)
```
and then rebuild
```
cd ORB_SLAM2/Examples/ROS/ORB_SLAM2
rm -r build
cd ORB_SLAM2
./build_ros.sh
```
finally, it runs well
Hope to help you