raulmur/ORB_SLAM2 #946: Time stamps are not used in motion model part of tracking

s271, 2020-10-07:
"motion model" used in tracking is just current frame in local coordinate of last frame.
Both carry time stamps, but the way it's implemented time stamps are ignored
```
mVelocity = mCurrentFrame.mTcw*LastTwc;
```
Shouldn't mVelocity be converted to Quaternion and Translation and both appropriately scaled by time delta according to time stamps?

xvzezi, 2020-10-21:
It is using a simple constant speed motion model, and what is done here also presumes the fps is stable.
```c++
mCurrentFrame.SetPose(mVel * mLastFrame.mTcw);
```
If one wants to rescale that, i believe do it in the se(3) space is a better idea.
The time delta is small, for a simple model, and a system not suited for fast speed, there should be small differences(not tested)

s271, 2020-10-25:
It seems you are right. I tested with slerp + translation interpolation, didn't see noticeable difference

closed 2020-10-25 by s271