raulmur/ORB_SLAM2 #950: Change displaysize of the Current Frame window?

JensAltst, 2020-10-26:
Hey I'm having trouble with viewing the current frame in my Linux VM.

My phone records videos/pictures in Full HD 1080px and the Viewer will always clip out of my screen. Is there an easy way to change the size of the window?

martinakos, 2021-01-19:
@JensAltst I'm looking for the same feature. Have you found a way of reducing the size of the current frame?

martinakos, 2021-01-19:
I found an easy way of resizing the current frame window. 
search for: 
`cv::namedWindow("ORB-SLAM3: Current Frame")`
in Viewer.cc
and replace with:
`cv::namedWindow("ORB-SLAM3: Current Frame", cv::WINDOW_NORMAL);`
then rebuild and the window will be resizable.

JensAltst, 2021-01-21:
Thanks that helped a lot!

closed 2021-01-21 by JensAltst