raulmur/ORB_SLAM2 #952: Why need ORBvoc.txt?

craftindo, 2020-11-10:
When I create an empty txt file, the code exited -1. Why?

Does that mean, when I run a brand new video, it will not work?
I thought building vocabulary of the scenery requires offline building database. If so, how does it work?

melhashash, 2020-12-02:
This file is required, you cannot create an empty one. It contains the vocabulary needed to run BoW, and It was created in a preprocessing step so you can use it on the fly while running ORB-SLAM.

craftindo, 2020-12-06:
thanks Melhashash, now I understand.

govindak-umd, 2021-04-05:
Please look into how the Bag of words approach works. It is an essential component to loop closure and helps complete your map. Hope that helps.