raulmur/ORB_SLAM2 #962: How to generate ORBvoc.txt?

liubamboo, 2020-12-10:
Thanks for your great work. I want to know how to create ORBvoc.txt. I want to generate ORBvoc.txt independently. Can you help me?

Gautham-JS, 2021-02-03:
Hello there, 

ORBvoc.txt is essentially a database of previously seen feature descriptors stored in a formatted file for fast place recognition for triggering loop closure instead of matching features one by one against all images.This type of place recognition is done using a hierarchical tree to create a bag of words representation of the image feature space.

This vocabulary is generated by using [DBoW2](https://github.com/dorian3d/DBoW2). You can read the paper at http://doriangalvez.com/papers/GalvezTRO12.pdf . In this repo, `./demo` shows a minimal example on creating vocabulary files.

In the context of SLAM and Loop closure detection with it, [This will be helpful](https://github.com/dorian3d/DLoopDetector).  

Also DBoW2 generates the vocabulary file in .yaml format and saves it as  `.yml.gz` format. but like you asked ORB SLAM2 uses `.txt` file, thats because the DBoW2 version provided with ORB SLAM2 in ./Thirdparty folder is slightly modified to support `.txt` files. [check out the source here](https://github.com/raulmur/ORB_SLAM2/blob/master/Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h).

Hope it helps.
G.

SnowCarter, 2021-02-15:
Hi, I can't find the './demo' folder. Could you please check it again?

Gautham-JS, 2021-02-16:
Hey @SnowCarter , 
The `./demo` folder i was refrencing to is from the original repo for DBoW2 : https://github.com/dorian3d/DBoW2

SnowCarter, 2021-02-17:
Great, thank you.

SnowCarter, 2021-02-17:
Please forgive my stupid question. If I create a vocabulary by using DBOW3. Can I apply it here (DBOW2)?

DanyPell, 2024-08-17:
@SnowCarter I do wonder if it would be better to use DBoW3, or FBoW (which DBoW3 repo says it's better): https://github.com/rmsalinas/fbow