raulmur/ORB_SLAM2 #965: mapping problem in simulation

mmahdavian, 2020-12-17:
Hi. I am trying to make orb-slam map in habitat simulation with replica dataset. So far I have generated path and got images and made rosbag from them. ORB-SLAM tries to make a map from the published images. The path is a C shaped motion. for first parts of it the map is really good, but after rotations the mapping doesn't work properly. Does anyone know how can I fix it? I have attached my map and habitat.yaml file and a ground truth showing the correct path.
[habitat.yaml.zip](https://github.com/raulmur/ORB_SLAM2/files/5706963/habitat.yaml.zip)

[image]

[image]

mmahdavian, 2020-12-17:
I increased the image size to 720x720 and modified habitat.yaml file based on it and the problem is solved

closed 2020-12-17 by mmahdavian

jediofgever, 2020-12-30:
what was the image size before  720x720 ? , I am even doing using 1280x720 but still same issue