raulmur/ORB_SLAM2 #969: Compiling finished BUT cannot excute？？？

Zaheng, 2020-12-24:
[image]

Zaheng, 2020-12-24:
It doesn't work at all！

Zaheng, 2020-12-24:
@raulmur help please!

AnhTuDo1998, 2021-01-11:
Any error message printed out when you try executing it ?

WaterHyacinthInNANHU, 2021-02-06:
I met a similar issue: nothing just happened when I execute
`./Examples/Monocular/mono_euroc Vocabulary/ORBvoc.txt Examples/Monocular/EuRoC.yaml ./dataset/mav0/cam0/data Examples/Monocular/EuRoC_TimeStamps/SEQUENCE.txt `
No error message, just nothing printed but hung up here.
Please help~

WaterHyacinthInNANHU, 2021-02-06:
> I met a similar issue: nothing just happened when I execute
> `./Examples/Monocular/mono_euroc Vocabulary/ORBvoc.txt Examples/Monocular/EuRoC.yaml ./dataset/mav0/cam0/data Examples/Monocular/EuRoC_TimeStamps/SEQUENCE.txt `
> No error message, just nothing printed but hung up here.
> Please help~

Stupid mistake...I forgot to replace `SEQUENCE.txt`. Thanks.

monsterbai, 2021-12-22:
same question。 have you solved this question yet?

cheslee-z, 2023-04-04:
I also made the stupid mistake