raulmur/ORB_SLAM2 #975: "Track lost soon after initialisation, reseting..." keeps happening

Amer-Alhamvi, 2021-02-09:
So I have this problem with different OpenCV versions, and on 2 different platforms. I have boiled it to the following:
The while loop in ORBmatcher.cc in line ~180 is always false since the first part "KFit != KFend" is stuck on false.
although the second condition "Fit != Fend" toggles no problem.

Thus the return value nmatches = 0, thus the boolean bOK is set to false in line 310 of Tracking.cc and the mState=LOST.

This is how far I could investigate, a newbie here sorry :)
If any one knows the solution or have faced a similar problem before I would appreciate your feedback.

thebirdgr, 2023-03-14:
Hi @Amer-Alhamvi Did you manage to solve it?

withnoerr, 2024-07-12:
> Hi @Amer-Alhamvi Did you manage to solve it?

Did you solve it? I have the same problem. The camera is zed2i