raulmur/ORB_SLAM2 #984: ORB_SLAM2 camera poses + COLMAP reconstruction

lkosh, 2021-03-24:
Hello,
I'm trying to run colmap dense reconstruction with camera poses pre-computed with ORB-SLAM2. I follow the instructions from here:https://colmap.github.io/faq.html?highlight=two%20view%20tracks#reconstruct-sparse-dense-model-from-known-camera-poses
To compute camera poses I use rgbd_tum.cc on tum1 sequence, the trajectory error compared to ground truth is quite small compared to the path reconstructed with COLMAP. 

I'm aware that world coordinate system used in COLMAP is different from conventional one, so I apply these transformations to poses outputted by ORB_SLAM2 (qx, qy, qz, qw are quaternions from CameraTrajectory.txt , tx, ty, tz - translations from ORB_SLAM2)
r = R.from_quat([qx, qy, qz, qw]).as_dcm()
r = r.T
r[0, 1] *= -1.0
r[0, 2] *= -1.0
r[1, 0] *= -1.0
r[2, 0] *= -1.0
qx, qy, qz, qw = R.from_dcm(r).as_quat()
ty *= -1
tz *= -1

However I found that dense reconstruction quality is much better without the pre-computed poses, whereas with the given poses colmap can reconstruct fewer points.

Here's the one of the depth maps of TUM1 sequence that my pipeline produces:
[image]
[image]

Has anyone else tried this orb-slam + colmap pipeline? Could someone explain what is the fundamental difference between camera poses that they use?

Mateo-tech, 2022-01-18:
Hey @lkosh, any success with this?

JunsukLee, 2022-07-06:
I have read related papers.
It is said to be advantageous for 3D reconstruction because the camera pose of ORB_SALM2 is much more accurate than the camera pose calculated by the COLMAP.

Cat-Lover-Yes, 2023-07-01:
Hai, have you solved this problem? @lkosh @mateomd-dev

Oliverbihop, 2024-01-09:
> I'm aware that world coordinate system used in COLMAP is different from conventional one, so I apply these transformations to poses outputted by ORB_SLAM2 (qx, qy, qz, qw are quaternions from CameraTrajectory.txt , tx, ty, tz - translations from ORB_SLAM2)

Hello, Could you please provide the tutorial and code of this