raulmur/ORB_SLAM2 #989: when I run build_ros.h, it has something wrong and shows undefined reference to `cv::FileNode::operator int() const'

cz-x, 2021-04-15:
The edition of ubuntu is 18.04 and opencv is 3.3.1.This is the specific error message:
/usr/bin/ld: warning: libopencv_imgproc.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_imgproc.so.3.3
/usr/bin/ld: warning: libopencv_core.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_core.so.3.3
CMakeFiles/MonoAR.dir/src/AR/ros_mono_ar.cc.o: In function `main':
ros_mono_ar.cc:(.text.startup+0xe88): undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
ros_mono_ar.cc:(.text.startup+0xecc): undefined reference to `cv::FileNode::operator int() const'
ros_mono_ar.cc:(.text.startup+0xef8): undefined reference to `cv::FileNode::operator float() const'
ros_mono_ar.cc:(.text.startup+0xf2c): undefined reference to `cv::FileNode::operator float() const'
ros_mono_ar.cc:(.text.startup+0xf4c): undefined reference to `cv::FileNode::operator float() const'
ros_mono_ar.cc:(.text.startup+0xf6c): undefined reference to `cv::FileNode::operator float() const'
ros_mono_ar.cc:(.text.startup+0xf8c): undefined reference to `cv::FileNode::operator float() const'
CMakeFiles/MonoAR.dir/src/AR/ros_mono_ar.cc.o:ros_mono_ar.cc:(.text.startup+0x1098): more undefined references to `cv::FileNode::operator float() const' follow
CMakeFiles/MonoAR.dir/src/AR/ViewerAR.cc.o: In function `ORB_SLAM2::ExpSO3(float const&, float const&, float const&)':

HilbertXu, 2021-07-23:
I intend to try ORB on Nvidia Jetson Xavier, but got the same error. Have you sovled this error?

Blossom-ASL, 2021-11-30:
@HilbertXu 
Have you solved this error?