raulmur/ORB_SLAM2 #990: running build_ros.h fails on Ubuntu 20.04(ros noetic)

IradNuriel, 2021-04-26:
HI, I'm having issues running build_ros.sh on ubuntu 20.04, this is the [CMakeLists.txt](https://github.com/raulmur/ORB_SLAM2/files/6375696/CMakeLists.txt) 
and this is the output of build_ros.sh:
`Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Warning (dev) at /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:155 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
Call Stack (most recent call first):
  CMakeLists.txt:4 (rosbuild_init)
This warning is for project developers.  Use -Wno-dev to suppress it.

[rosbuild] Building package ORB_SLAM2
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:366 (cmake_policy):
  The OLD behavior for policy CMP0037 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:4 (rosbuild_init)

-- Using CATKIN_DEVEL_PREFIX: /home/iradnuriel/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.5", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/iradnuriel/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.5") 
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.9
-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[rosbuild] Including /opt/ros/noetic/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/noetic/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/noetic/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/iradnuriel/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
[ 10%] Linking CXX executable ../Mono_Load
[ 20%] Linking CXX executable ../Stereo
[ 30%] Linking CXX executable ../Mono_Save
[ 50%] Built target RGBD
[ 70%] Built target Mono
/usr/bin/ld: CMakeFiles/Mono_Load.dir/src/ros_mono_load.cc.o: undefined reference to symbol '_ZN2cv7imwriteERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_11_InputArrayERKSt6vectorIiSaIiEE'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.4.2.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/Mono_Load.dir/build.make:221: ../Mono_Load] Error 1
make[1]: *** [CMakeFiles/Makefile2:786: CMakeFiles/Mono_Load.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/Mono_Save.dir/src/ros_mono_save.cc.o: undefined reference to symbol '_ZN2cv7imwriteERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKNS_11_InputArrayERKSt6vectorIiSaIiEE'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.4.2.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/Mono_Save.dir/build.make:221: ../Mono_Save] Error 1
make[1]: *** [CMakeFiles/Makefile2:704: CMakeFiles/Mono_Save.dir/all] Error 2
/usr/bin/ld: warning: libopencv_imgproc.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_imgproc.so.3.4
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /opt/ros/noetic/lib/libcv_bridge.so, may conflict with libopencv_core.so.3.4
/usr/bin/ld: CMakeFiles/Stereo.dir/src/ros_stereo.cc.o: undefined reference to symbol '_ZNK2cv8FileNodecviEv'
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libopencv_core.so.4.2.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/Stereo.dir/build.make:221: ../Stereo] Error 1
make[1]: *** [CMakeFiles/Makefile2:542: CMakeFiles/Stereo.dir/all] Error 2
make: *** [Makefile:130: all] Error 2`

I saw someone had a similar problem with build_ros.sh in ORB_SLAM3, but he found no solutions to that.
Hope you can help me find a solution to this problem.

arjunskumar, 2021-04-30:
@IradNuriel you were able to run `./build.sh` successfully?

IradNuriel, 2021-04-30:
Yes.

arjunskumar, 2021-05-03:
@IradNuriel Thanks

owenwijaya177, 2021-07-28:
Hi I'm also facing this problem. Does anyone have a solution?

IradNuriel, 2021-07-30:
The way I solved it was to copy the libopencv_core.so, libopencv_imgcodecs.so, libopencv_imgproc.so files into the lib folder(along with the .4.2.0 file for each one) and linked them directly through the CMakeLists.txt

closed 2021-07-30 by IradNuriel

Taomin00, 2022-03-17:
> 

Thank you. I have successfully solved this problem through your method

MoonSheepJx, 2022-04-08:
> > 
> 
> Thank you. I have successfully solved this problem through your method

I read the solution, but I still don't understand, can you give me a specific solution?
thanks

SBaokun, 2022-04-09:
hello，i have the same probles as you,can you describe the solution in detail?thanks

MoonSheepJx, 2022-04-09:
> hello，i have the same probles as you,can you describe the solution in detail?thanks

i have solved it, remove existing bridge (ros’s), download new bridge and alter the bridge cmakelists to find opencv which version you used, then cmake..  make sudo make install ,finally copy the new bridge to the /opt/ros/noetic/ (The path of the bridge just deleted )

nopnop2002, 2023-01-17:
@SBaokun 

Perhaps you are using Ubuntu 20.04.
build_ros.sh needs bridge.
When using Ubuntu20.04, it is necessary to replace the bridge.

- Ubunto20.04
```
$ ls -l /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so*
lrwxrwxrwx 1 root root      24  2月 18  2020 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so -> libopencv_imgproc.so.4.2
lrwxrwxrwx 1 root root      26  2月 18  2020 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2 -> libopencv_imgproc.so.4.2.0
-rw-r--r-- 1 root root 5266136  2月 18  2020 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.4.2.0
```

- Ubuntu 18.04
```
$ ls -l /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so*
lrwxrwxrwx 1 root root      24  9月 20  2018 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so -> libopencv_imgproc.so.3.2
lrwxrwxrwx 1 root root      26  9月 20  2018 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.3.2 -> libopencv_imgproc.so.3.2.0
-rw-r--r-- 1 root root 2862992  9月 20  2018 /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.3.2.0
```

yaoxingzhi, 2023-03-28:
> > hello，i have the same probles as you,can you describe the solution in detail?thanks
> 
> i have solved it, remove existing bridge (ros’s), download new bridge and alter the bridge cmakelists to find opencv which version you used, then cmake.. make sudo make install ,finally copy the new bridge to the /opt/ros/noetic/share (The path of the bridge just deleted )

你好，请问你具体怎样解决的呢？谢谢

irsyadtc, 2023-04-14:
> The way I solved it was to copy the libopencv_core.so, libopencv_imgcodecs.so, libopencv_imgproc.so files into the lib folder(along with the .4.2.0 file for each one) and linked them directly through the CMakeLists.txt

Questions:
- Instead of copy, can I symlink?
- How do you linked them directly through CMakeLists.txt? Could you please show the command & at which lines?

rocker12121, 2023-09-21:
Hello. I just solved this problem in the simplest way. You just need to replace the version of OpenCV in CMakeLists.txt in ORB_SLAM2/Examples/ROS/ORB_SLAM2/ from 3.0 to 4.2.

IradNuriel, 2023-09-21:
You are correct, but since opencv is not known for the best backward-compatibility I did not do it at the time

ashwinjoseph95, 2024-08-23:
> Hello. I just solved this problem in the simplest way. You just need to replace the version of OpenCV in CMakeLists.txt in ORB_SLAM2/Examples/ROS/ORB_SLAM2/ from 3.0 to 4.2.

But there are too many changes needed to be made in hte orbslam2 code to replace opencv related lines, are you aware of any orbslam2 codebase for which these changes are made already?

rocker12121, 2024-08-23:
> But there are too many changes needed to be made in hte orbslam2 code to replace opencv related lines, are you aware of any orbslam2 codebase for which these changes are made already?

I stopped working on this a long time ago. But if my memory is correct, you need to make changes only if your opencv version is not compatible. As long as you use the compatible opencv (4.2), just change the version to 4.2 in cmakelist.

ashwinjoseph95, 2024-08-23:
> I stopped working on this a long time ago. But if my memory is correct, you need to make changes only if your opencv version is not compatible. As long as you use the compatible opencv (4.2), just [...]

Oh ok thanks for the reply, when I tried to build opencv 4.2 from scratch and change the version in cmakelist it is throwing many opencv incompatibilty issues.