raulmur/ORB_SLAM2 #991: Unable to building package ORB_SLAM2

saleelirenyun, 2021-04-29:
Hey guys, when it comes to 7. ROS Examples, after I have reset the ROS Path, I get the following warning.

[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from syntax check of ORB_SLAM2/manifest.xml
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'roslib'
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:77 (message):
  [rosbuild] Syntax check of ORB_SLAM2/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/nxf70322/VSLAM_RYL/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

What should I do?
Thank advanced!

DaDDong, 2021-11-12:
Hello, I have the same problem. Do you solved it?

ShenhavLazarovich, 2021-12-14:
Had the same problem, did you solved it?

DaDDong, 2021-12-14:
I was using Ubuntu 20.04, but later installed 18.04 and ran it successfully in 18.04