raulmur/ORB_SLAM2 #999: KeyFrames are getting closer as ORB SLAM2 continues

deeprun, 2021-06-29:
Hi, I am running with my own data and I found the keyframes in the Map viewer are getting denser (closer) to each other. As shown in the picture attached . Why is that?

[image]

AlejandroSilvestri, 2021-07-29:
@deeprun 

Hard to tell.  There can be a lot of reasons.

First, I noticed you are turning right, that can explain the need of more keyframes while you discover previously unseen scene.

Second, there is a keyframe culling procedure that will eliminate redundant keyframes.  May be there are keyframes waiting to be erased.

Third, I saw such a behaviour with bad calibration.  I mean very bad calibration.  If you can see a straight chessboard near the vertices of you undistorted images, then you have a good calibration.  It is always a good idea to visually verify to calibration.

Fourth, may be your scale is shrinking, and you need to walk a shorter real distance to get a new keyframe.  Often closing a loop solves this.

And there may be a lot of other reasons.

M3000j, 2022-03-27:
how to solve or prevent it?

AlejandroSilvestri, 2022-03-27:
@M3000j 

1. Improve your calibration, specially your distortion coefficients
2. Make a loop
3. Wait to automatic keyframe culling take place