raulmur/ORB_SLAM2 #1000: Opencv 4 support

deeprun, 2021-07-04:
I am trying to change the code so I can use opencv 4. Most of the changes are straightforward. But PnPsolver.cc used a lot of opencv c API which is hard to find replacement in opencv 4. Have anyone transferred this code to opencv4?

crmaxj, 2021-07-14:
Check out this PR's branch, it should just work for you! https://github.com/raulmur/ORB_SLAM2/pull/970

Mechazo11, 2022-10-11:
> Check out this PR's branch, it should just work for you! #970

It doesn't however I got this project to compile https://github.com/Mateo-tech/orb-slam2. Will comment back when I get a custom dataset that I know works with ORB_SLAM3 to run with this version of ORB_SLAM2