raulmur/ORB_SLAM2 #1007: Need Help

gcc-hub, 2021-07-21:
need help，which rule can create this file libORB_SLAM2.so？

[image]

AlejandroSilvestri, 2021-08-18:
Hi @gcc-hub 

I believe [this one](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/CMakeLists.txt#L51) is the rule you're looking for.

`add_library(${PROJECT_NAME} SHARED`

gcc-hub, 2021-08-18:
> Hi @gcc-hub
> 
> I believe [this one](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/CMakeLists.txt#L51) is the rule you're looking for.
> 
> `add_library(${PROJECT_NAME} SHARED`

gcc-hub, 2021-08-18:
@

> Hi @gcc-hub
> 
> I believe [this one](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/CMakeLists.txt#L51) is the rule you're looking for.
> 
> `add_library(${PROJECT_NAME} SHARED`

gcc-hub, 2021-08-18:
> @
> 
> > Hi @gcc-hub
> > I believe [this one](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/CMakeLists.txt#L51) is the rule you're looking for.
> > `add_library(${PROJECT_NAME} SHARED`

gcc-hub, 2021-08-18:
Thanks for your advice, I think this is a key to solving the problem

IvySong10086, 2022-03-03:
Do you solve this problem?