raulmur/ORB_SLAM2 #1016: Use a built-in camera

zsoltszakaly, 2021-09-10:
Apologies, I am new to the ROS world, so it might be obvious for many, but after days of google I could not figure it out:

I have ORB_SLAM2 up and running. Examples are working fine. I want to use the built-in camera for on-line processing.
1. I started `roscore` in one terminal. Up and running.
2. In another terminal I started `rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml`. I know that the parameter files might not be suitable for this run, but that shall come later. It looks everything runs fine, I have the Map Viewer and the Current Frame windows open. (If I do not do Step 1, I get an error that it cannot connect to the master, but after a Ctrl-C it still gets to this point, so I wonder what it uses Step 1, but anyway, I have it running).

The problem is that the Current Frame is "WAITING FOR IMAGES". I read somewhere that I need to rosbag /camera/data. So in a third terminal I have `rosbag record /camera/data`. I am not sure, it is needed (I would think that it is done internally in Step 2 anyway), but it works. I see the /camera/data in `rostopic list`. Still nothing, my camera is actually not even switched on.

So my question: How to use the buil-in camera, just switching it on and letting it stream the frames to ORB_SLAM2?

Thanks,

Faizi344, 2022-08-05:
@zsoltszakaly can u tell me how  u run the examples

zsoltszakaly, 2022-08-08:
Honestly, I do not remember, but the good news that somehow I could make it run. Still with one laptop built-in camera the map generated was very far from accurate. I know I should use a larger resolution, or even more, two cameras, best if combined with GPS (there are some projects for that too), but I had no time to deal with it for a long time.

thebirdgr, 2023-03-14:
You have quite a lot of things missing. Just running 
`rosbag record /camera/data` won't help. You need to have a bag file like the example and run the bag file. .A ROSBag file is a recording of topics during a particular instance. In the case of the euroc stereo example, it was the imu, rgb, depth images and some other stuff. Your `Mono` code will subscribe to certain topic names as stated by the documentation, `/camera/data`. 

I see you're trying to record a bag file. Once that is recorded, you can use `rosbag play file.bag`/