raulmur/ORB_SLAM2 #1018: ./build.sh  error   when i run this  it happened ,im a rookie in this somebody can help me?

yingxinli, 2021-09-23:
[ 81%] Linking CXX executable ../Examples/Stereo/stereo_kitti
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘ORB_SLAM2::MapDrawer::MapDrawer(ORB_SLAM2::Map*, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/mono_kitti.dir/build.make:159: recipe for target '../Examples/Monocular/mono_kitti' failed
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
CMakeFiles/Makefile2:224: recipe for target 'CMakeFiles/mono_kitti.dir/all' failed
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘ORB_SLAM2::MapDrawer::MapDrawer(ORB_SLAM2::Map*, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/rgbd_tum.dir/build.make:159: recipe for target '../Examples/RGB-D/rgbd_tum' failed
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
CMakeFiles/Makefile2:120: recipe for target 'CMakeFiles/rgbd_tum.dir/all' failed
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘ORB_SLAM2::MapDrawer::MapDrawer(ORB_SLAM2::Map*, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/mono_tum.dir/build.make:159: recipe for target '../Examples/Monocular/mono_tum' failed
make[2]: *** [../Examples/Monocular/mono_tum] Error 1
CMakeFiles/Makefile2:250: recipe for target 'CMakeFiles/mono_tum.dir/all' failed
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘ORB_SLAM2::MapDrawer::MapDrawer(ORB_SLAM2::Map*, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_kitti.dir/build.make:159: recipe for target '../Examples/Stereo/stereo_kitti' failed
make[2]: *** [../Examples/Stereo/stereo_kitti] Error 1
CMakeFiles/Makefile2:172: recipe for target 'CMakeFiles/stereo_kitti.dir/all' failed
make[1]: *** [CMakeFiles/stereo_kitti.dir/all] Error 2
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘ORB_SLAM2::MapDrawer::MapDrawer(ORB_SLAM2::Map*, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/mono_euroc.dir/build.make:159: recipe for target '../Examples/Monocular/mono_euroc' failed
make[2]: *** [../Examples/Monocular/mono_euroc] Error 1
CMakeFiles/Makefile2:94: recipe for target 'CMakeFiles/mono_euroc.dir/all' failed
make[1]: *** [CMakeFiles/mono_euroc.dir/all] Error 2
../lib/libORB_SLAM2.so：对‘pangolin::Split(std::string const&, char)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreatePanel(std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::fromString(cv::Mat&, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::BindToContext(std::string)’未定义的引用
../lib/libORB_SLAM2.so：对‘ORB_SLAM2::MapDrawer::MapDrawer(ORB_SLAM2::Map*, std::string const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘DBoW2::FORB::toString(cv::Mat const&)’未定义的引用
../lib/libORB_SLAM2.so：对‘pangolin::CreateWindowAndBind(std::string, int, int, pangolin::Params const&)’未定义的引用
collect2: error: ld returned 1 exit status
CMakeFiles/stereo_euroc.dir/build.make:159: recipe for target '../Examples/Stereo/stereo_euroc' failed
make[2]: *** [../Examples/Stereo/stereo_euroc] Error 1
CMakeFiles/Makefile2:198: recipe for target 'CMakeFiles/stereo_euroc.dir/all' failed
make[1]: *** [CMakeFiles/stereo_euroc.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2

LMWafer, 2021-09-25:
I had a similar problem and added 
```C
#include<unistd.h>
```
into *include/System.h* before running
```bash
./build.sh
```
in *ORB_SLAM2/*. 

As *System.h* is probably one of the first files to be called, adding the code will apply its modifications to all the other headers called within/after it. 

If you still have this problem, add the *include* to all the *.cc* files mentioned in 
```bash
CMakeFiles/<filedir/filename>make:159: recipe for target '<filedir/filename.cc>`
```
-like lines and retry
```bash
rm -rf build && ./build.sh
```
until you have no more files do modify !

yingxinli, 2021-09-26:
> I had a similar problem and added
> 
> ```c
> #include<unistd.h>
> ```
> 
> into _include/System.h_ before running
> 
> ```shell
> ./build.sh
> ```
> 
> in _ORB_SLAM2/_.
> 
> As _System.h_ is probably one of the first files to be called, adding the code will apply its modifications to all the other headers called within/after it.
> 
> If you still have this problem, add the _include_ to all the _.cc_ files mentioned in
> 
> ```shell
> CMakeFiles/<filedir/filename>make:159: recipe for target '<filedir/filename.cc>`
> ```
> 
> -like lines and retry
> 
> ```shell
> rm -rf build && ./build.sh
> ```
> 
> until you have no more files do modify !

thank you for you help! I have already added and I found that my Pangolin version is the latest version that maybe have some bug or mistake. So i change Pangolin to the old version and succeed  "build.sh" . Thank you again!

closed 2021-09-26 by yingxinli

IncludeMathH, 2024-01-09:
I had the same problem!