raulmur/ORB_SLAM2 #1022: error when sudo ./build_ros.sh(ros-noetic)

DaDDong, 2021-11-10:
I have successfully compiled build.sh, but when I started to execute build_ros.sh, an error is reported:
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/noetic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from syntax check of ORB_SLAM2/manifest.xml
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/__init__.py", line 50, in <module>
    from roslib.launcher import load_manifest  # noqa: F401
  File "/opt/ros/noetic/lib/python3/dist-packages/roslib/launcher.py", line 42, in <module>
    import rospkg
ImportError: No module named rospkg
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:77 (message):
  [rosbuild] Syntax check of ORB_SLAM2/manifest.xml failed; aborting
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:174 (_rosbuild_check_manifest)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/yxd/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** 没有指明目标并且找不到 makefile。 停止。

Miha22, 2022-03-22:
`project(package_name)`

yaoxingzhi, 2023-03-27:
i have the same error .have you slove the problem?