raulmur/ORB_SLAM2 #1026: Build failed with opencv 5++

ShenhavLazarovich, 2021-12-14:
Hi
Can you tell me the current versions of each dependency(And Ubuntu version) which you have a successful  build with? 
I've tried too many time and each time get a different error .

Thank you!

Mateo-tech, 2021-12-28:
Hey, I've spent quite a lot of time trying to make it work with OpenCV 4.X (4.5.4 to be specific, Pangolin 0.6 and C++14 compiler).
I shared the updated code for ORB_SLAM2:
https://github.com/Mateo-tech/orb-slam2

You'll have to get the OpenCV and Pangolin yourself (I built it from source)

jokolaco, 2022-10-27:
@ mateomd-dev
I can confirm that your update is working. 
At least I am working since a few month with your updated OrbSlam without problems.
Thanks!

supermice, 2022-10-27:
邮件已收到！ ***@***.***