raulmur/ORB_SLAM2 #1027: build_ros.sh giving this error: recipe for target 'CMakeFiles/Stereo.dir/src/ros_stereo.cpp.o' failed

low-carbon, 2021-12-20:
[ 66%] Linking CXX executable ../Mono
c++: internal compiler error: 段错误 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
CMakeFiles/RGBD.dir/build.make:118: recipe for target 'CMakeFiles/RGBD.dir/src/ros_rgbd.cpp.o' failed
make[2]: *** [CMakeFiles/RGBD.dir/src/ros_rgbd.cpp.o] Error 4
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/RGBD.dir/all' failed
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 66%] Built target Mono
c++: internal compiler error: 段错误 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
CMakeFiles/Stereo.dir/build.make:118: recipe for target 'CMakeFiles/Stereo.dir/src/ros_stereo.cpp.o' failed
make[2]: *** [CMakeFiles/Stereo.dir/src/ros_stereo.cpp.o] Error 4
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/Stereo.dir/all' failed
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
[ 77%] Linking CXX executable ../MonoAR
[ 77%] Built target MonoAR
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

samuelyutt, 2022-05-26:
Did you solve this problem?

low-carbon, 2022-05-30:
> Did you solve this problem?

I once got this error installing orb-slam2 on an arm system, then I reinstalled the arm system and solved the problem