raulmur/ORB_SLAM2 #1033: Stuck or crash at using ORB_SLAM2 Mono

ShamanStalker, 2022-02-25:
Good day
I used `rosrun ORB_SLAM2 Mono ~/ORB_SLAM2/Vocabulary/ORBvoc.txt ~/ORB_SLAM2/Examples/Monocular/our.yaml /camera/image_raw:=/usb_cam/image_raw`
The viewport opens and I see lots of green lines and points, but the process stuck at `TRYING TO INITIALIZE`. If I change `ORBextractor.nFeatures` in `our.yaml` from 1000 to 2000, it will case nothing. If `ORBextractor.nFeatures` more then 4000-5000, then ORB_SLAM2 crashes with `New Map created with 109 points Segmentation fault (core dumped)`, when I'm trying to look around with my camera
Eigen version is 3.2.10
OpenCV version is 2.4.8
ORB_SLAM2 version is the latest
Pangolin version is 0.5
I also add #include <unistd.h> in each necessary file as in other answers
Cmake version is 3.22.2
Ubuntu version is 14.04.6 LTS
ROS Indigo installed
What else?..
I am somewhat n00b in Linux, so sorry if the question is obvious
Also sorry if there is an answer, I did not find it yet.
Thanx

javadan, 2022-03-09:
Also just got it 'TRYING TO INITIALIZE' for mono, though I'm in ros2 foxy, and on a jetson, running a port of this codebase.  
My green lines are very sparse, haven't seen any points yet. 

I'll reply if I find a solution.

EndlessLoops, 2023-03-10:
Did you solve the problem?I have the same problem in ros2 humble.

> 虽然我在 ros2 foxy 和 jetson 上运行这个代码库的一个端口，但也刚刚得到它的“尝试初始化”用于单声道。 我的绿线很稀疏，还没有看到任何点。
> 
> 如果我找到解决方案，我会回复。