raulmur/ORB_SLAM2 #1038: /usr/bin/ld: cannot find -lEigen3::Eigen

Xue21, 2022-04-10:
chmod +x build.sh
./build.sh
Then it is wrong as below.
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubutu18/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:310 (if):
  if given arguments:

    "TRUE"

  An argument named "TRUE" appears in a conditional statement.  Policy
  CMP0012 is not set: if() recognizes numbers and boolean constants.  Run
  "cmake --help-policy CMP0012" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindOpenMP.cmake:425 (_OPENMP_GET_SPEC_DATE)
  CMakeLists.txt:47 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubutu18/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
CMake Error at CMakeLists.txt:40 (find_package):
  Found package configuration file:

    /home/ubutu18/Pangolin/build/PangolinConfig.cmake

  but it set Pangolin_FOUND to FALSE so package "Pangolin" is considered to
  be NOT FOUND.  Reason given by package:

  Pangolin could not be found because dependency Eigen3 could not be found.

-- Configuring incomplete, errors occurred!
See also "/home/ubutu18/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubutu18/ORB_SLAM2/build/CMakeFiles/CMakeError.log".
[  3%] Linking CXX shared library ../lib/libORB_SLAM2.so
/usr/bin/ld: cannot find -lEigen3::Eigen
collect2: error: ld returned 1 exit status
CMakeFiles/ORB_SLAM2.dir/build.make:624: recipe for target '../lib/libORB_SLAM2.so' failed
make[2]: *** [../lib/libORB_SLAM2.so] Error 1
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

Orange2219, 2022-05-20:
please install pangolin

Hello-Water, 2022-08-31:
if you have installed the "Prerequisites", please delete the document "FindEigen3.cmake" in "cmake_modules" and add your path where eigen3 installed, i.e., "/usr/include/eigen3"
Or
replace the document "FindEigen3.cmake" with your own file "FindEigen3.cmake".

9a3eedi, 2022-08-31:
Thank you @Hello-Water and @gitbug1949 , I was finally able to build ORB-SLAMv3 on Ubuntu 18.04. 

What I did is install eigen from the package manager, install pangolin after building it from source (i.e. sudo make install), then I deleted FindEigen3.cmake, and in the CMakeLists.txt, I added "/usr/include/eigen3" to the include_directories() lines

IHKYoung, 2022-09-06:
just remove cmake_modules