raulmur/ORB_SLAM2 #1042: Build Error!

Summer2474, 2022-04-23:
ORB-SLAM3 works well in my computer, however, when I try to build ORB-SLAM2, I get some strange errors that I fail to solve!
If someone encounters a mistake similar to mine, please tell me how to fix it, and I don't find any effective solution now.

Configuring and building Thirdparty/DBoW2 ...
mkdir: 无法创建目录 “build”: 文件已存在
CMake Error: The current CMakeCache.txt directory /home/summer/ORB-SLAM2/Thirdparty/DBoW2/build/CMakeCache.txt is different than the directory /home/summer/文档/ORB-SLAM2/Thirdparty/DBoW2/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/summer/ORB-SLAM2/Thirdparty/DBoW2/CMakeLists.txt" does not match the source "/home/summer/文档/ORB-SLAM2/Thirdparty/DBoW2/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: 无法创建目录 “build”: 文件已存在
CMake Error: The current CMakeCache.txt directory /home/summer/ORB-SLAM2/Thirdparty/g2o/build/CMakeCache.txt is different than the directory /home/summer/文档/ORB-SLAM2/Thirdparty/g2o/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/summer/ORB-SLAM2/Thirdparty/g2o/CMakeLists.txt" does not match the source "/home/summer/文档/ORB-SLAM2/Thirdparty/g2o/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: 无法创建目录 “build”: 文件已存在
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
CMake Warning at CMakeLists.txt:110 (add_executable):
  Cannot generate a safe runtime search path for target mono_euroc because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpng16.so.16] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
    runtime library [libtiff.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:106 (add_executable):
  Cannot generate a safe runtime search path for target mono_kitti because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpng16.so.16] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
    runtime library [libtiff.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:102 (add_executable):
  Cannot generate a safe runtime search path for target mono_tum because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpng16.so.16] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
    runtime library [libtiff.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib

  Some of these libraries may not be found correctly.

CMake Warning at CMakeLists.txt:95 (add_executable):
  Cannot generate a safe runtime search path for target stereo_euroc because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libpng16.so.16] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/summer/anaconda3/lib
[... 189 lines cut ...]
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `wattr_on@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `resize_term@NCURSESW6_5.3.20021019'
make[2]: *** [CMakeFiles/stereo_kitti.dir/build.make:162：../Examples/Stereo/stereo_kitti] 错误 1
make[1]: *** [CMakeFiles/Makefile2:196：CMakeFiles/stereo_kitti.dir/all] 错误 2
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `getmouse@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `COLORS@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `wgetch@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `wmove@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined reference to `cairo_font_options_get_variations'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `initscr@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `waddnstr@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `wrefresh@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined reference to `cairo_tag_begin'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `wattr_on@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `getmouse@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `wgetch@NCURSESW6_5.1.20000708'
/usr/bin/ld: /lib/x86_64-linux-gnu/librsvg-2.so.2: undefined reference to `cairo_font_options_get_variations'
/usr/bin/ld: /lib/x86_64-linux-gnu/libcaca.so.0: undefined reference to `initscr@NCURSESW6_5.1.20000708'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/rgbd_tum.dir/build.make:162：../Examples/RGB-D/rgbd_tum] 错误 1
make[1]: *** [CMakeFiles/Makefile2:223：CMakeFiles/rgbd_tum.dir/all] 错误 2
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/mono_euroc.dir/build.make:162：../Examples/Monocular/mono_euroc] 错误 1
make[1]: *** [CMakeFiles/Makefile2:88：CMakeFiles/mono_euroc.dir/all] 错误 2
make: *** [Makefile:84：all] 错误 2

musalmaz, 2023-01-26:
Hello, did you solve this issue, I got the same problem. Any help would be appreciated.

pramitd, 2023-02-09:
Same problem. Do you have the solution. Kindly give a reply, thanks.

musalmaz, 2023-02-27:
I don't understand which way solved my problem but it possible solutions are:
1 - if you use conda,  run your command without a conda environment, even base., 
2 -  the reason for the problem may is opencv, resolve this.

sun-bz, 2024-03-22:
i also meet this question.Do you have the solution?

supermice, 2024-03-22:
邮件已收到！ ***@***.***

lx-r, 2024-03-22:
您好，来信已收到，谢谢！