raulmur/ORB_SLAM2 #1048: ./build_ros.sh error

tungtungyan, 2022-05-18:
I have the error when running ./build_ros.sh.
[rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is
  "/media/ytl/2TB/catkin_orb/src/ORB_SLAM2_modified/Examples/ROS/ORB_SLAM2".
  You should double-check your ROS_PACKAGE_PATH to ensure that packages are
  found in the correct precedence order.

However, my $ROS_PACKAGE_PATH is this.
ytl@ytl-System-Product-Name:~$ echo $ROS_PACKAGE_PATH
/opt/ros/melodic/share:/media/ytl/2TB/catkin_orb/src/ORB_SLAM3/Examples/ROS:PATH/ORB_SLAM2_modified/Examples/ROS
ytl@ytl-System-Product-Name:~$ echo $ROS_ROOT
/opt/ros/melodic/share/ros

My .bashrc file:
source /opt/ros/melodic/setup.bash
#export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM3/Examples/ROS
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:PATH/ORB_SLAM2_modified/Examples/ROS

Ubuntu 18.04

fikricanozgur, 2022-05-28:
I have the exact problem. Did you solve it?

zcjysustech, 2022-06-06:
Maybe "source ~/.bashrc" can be useful