raulmur/ORB_SLAM2 #1056: help me please

18392193281, 2022-09-06:
hi,I want to ask a question. that is how to come true a track issue.I need realize back vehicle track front vehicle by camera.But i dont know how to do?how to get pose and give to the code?if you know,please teach me.thank you.

Mechazo11, 2022-09-28:
@18392193281 Might wanna reword the question. How about breaking down what the steps you are trying to do? I have been working with ORB-SLAM 1 and 2 quite a while now since I am building a Python (Cython boosted) version for a different project. Maybe I can help if you can reword the question with more clarification.

18392193281, 2022-09-29:
> @18392193281 Might wanna reword the question. How about breaking down what the steps you are trying to do? I have been working with ORB-SLAM 1 and 2 quite a while now since I am building a Python [...]

gyhuui

Hardik7674, 2023-05-01:
> @18392193281 Might wanna reword the question. How about breaking down what the steps you are trying to do? I have been working with ORB-SLAM 1 and 2 quite a while now since I am building a Python [...]

how to generate the corresponding [yaw, x, z] value of ego car odometry using the orbslam2 library. please help me

Mechazo11, 2023-05-01:
@Hardik7674 since my response to @18392193281, I have switched over to using the C++ version of ORB SLAM3. I suggest u bump your project upto ORB SLAM3 first. 

Regarding your query, you have to clarify first something. Are you looking to find 6DOF pose (from which you can get the z, x and yaw) Frame by Frame or for each new KeyFrame? ORB SLAM3 is a Keyframe based VLSAM framework. I strongly suggest you read up on Frame and Keyframe concept and then rephrase your question.