raulmur/ORB_SLAM2 #1067: ROS2 support ?

zoldaten, 2022-12-25:
hello!
does ORB_SLAM2 support ROS2 ? or just ROS1?

if "yes" how to build nodes?
i receive this:
```
al@al:~/ORB_SLAM2$ ./build_ros.sh 
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /core/rosbuild/rosbuild.cmake

CMake Error at CMakeLists.txt:4 (rosbuild_init):
  Unknown CMake command "rosbuild_init".
```
i know that it is because i use colcon build instead catkin. but may be it could be fixed ?

PhilShaw852, 2023-08-25:
I have the same issue, did you ever fix it?

zoldaten, 2023-08-25:
hi @PhilShaw852  i moved to stella_vslam.

closed 2023-08-25 by zoldaten