raulmur/ORB_SLAM2 #1068: RB SLAM 2 error while executing mono_tum: no such file or directory

AhmedJabareen96, 2022-12-25:
I had successfully built ORB SLAM 2, and I want to run the Monocular example. I used the command:

`./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml /home/$USER/Downloads/rgbd_dataset_freiburg1_xyz`

to run the mono_tum executable on the given data set, but I get the following error:

`no such file or directory: ./Examples/Monocular/mono_tum`

What should I do ?

oguzhanilter, 2022-12-26:
Hi,

Be sure that 
- you correctly created the executable of _mono_tum_ . For that, please check the last part of the CMakeLists.txt to see if the 'add_executable' for the mono_tum is correct. 
- Please check if the executable is present in the directory of  <path_to_ORBSLAM2> /ORB_SLAM2/Examples/Monocular/
-  Please be sure that you are running the command in the correct directory: the commend you provided, should be run inside the  '<path_to_ORBSLAM2> /ORB_SLAM2'

I hope it helps

hhhtaotao, 2023-11-03:
Hello, I also have the same problem, did you solve the problem? Can you tell me how to fix it? Thank you so much