raulmur/ORB_SLAM2 #1070: an error when running command ./build.sh

ADgai234, 2023-01-10:
Ubuntu 20.04 Noetic
Hello,the following is the error.Could you please help me?
[ 65%] Linking CXX executable ../Examples/Monocular/mono_euroc
/usr/bin/ld: ../lib/libORB_SLAM2.so: undefined reference to `DBoW2::FORB::L'
/usr/bin/ld: ../lib/libORB_SLAM2.so: undefined reference to `DUtils::Random::RandomInt(int, int)'
/usr/bin/ld: ../lib/libORB_SLAM2.so: undefined reference to `DUtils::Random::SeedRandOnce()'
/usr/bin/ld: ../lib/libORB_SLAM2.so: undefined reference to `DUtils::Random::SeedRandOnce(int)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/mono_euroc.dir/build.make:121：../Examples/Monocular/mono_euroc] 错误 1
make[1]: *** [CMakeFiles/Makefile2:88：CMakeFiles/mono_euroc.dir/all] 错误 2
make: *** [Makefile:84：all] 错误 2

alam121, 2023-02-06:
The same error came for me. One wayout is to delete the build folder and build again:
mkdir build
cd build
cmake ..
make

This worked for me  after ./buid.sh gave error

mgrallos, 2023-03-04:
my error while running build.sh is 

-- Configuring incomplete, errors occurred!
See also "/home/mariah/trial_ws/src/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

irsyadtc, 2023-04-06:
I tried both `./build.sh` and @alam121 approaches but both resulted to this error: 
```shell
/home/irsyad2/Pangolin/components/pango_core/include/sigslot/signal.hpp:109:79: error: ‘decay_t’ is not a member of ‘std’; did you mean ‘decay’?
  109 | _weak_ptr_compatible_v = detail::is_weak_ptr_compatible<std::decay_t<P>>::value;
         |                                                              ^~~~~~~
         |                                                              decay

``` 
I built Pangolin from stevenlovegrove's repository at my home directory, master branch. Help please?

Ashwij3, 2023-04-25:
@irsyadtc Any updates on the issue? I am facing the same error.

irsyadtc, 2023-04-26:
Yes, see https://github.com/raulmur/ORB_SLAM2/issues/1015
Pangolin is currently using a higher version c++ compiler while ORB-SLAM2 uses the older version.
Try remove Pangolin and `git clone -b v0.5 https://github.com/stevenlovegrove/Pangolin.git` then cmake again