raulmur/ORB_SLAM2 #1084: ROS noetic ./build_ros.sh problems

bzofsv, 2023-05-21:
I'm trying to build ROS examples, but I keep getting this error. The other issues posts said to source setup.bash in .bashrc, but this file doesn't exist. I've tried all the other solutions that I can find, and none work.

My .bashrc:

source /home/user/ORB_SLAM2/Examples/ROS/devel/setup.bash
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/user/ORB_SLAM2/Examples/ROS

My ORB_SLAM2/Examples/ROS/CMakeLists.txt changes:

${Pangolin_LIBRARIES}
${PROJECT_SOURCE_DIR}/../../../Thirdparty/DBoW2/lib/libDBoW2.so
${PROJECT_SOURCE_DIR}/../../../Thirdparty/g2o/lib/libg2o.so
${PROJECT_SOURCE_DIR}/../../../lib/libORB_SLAM2.so
 -lboost_system
)

Error message:

[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from directory check: /opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/bzofsv/ORB_SLAM2/Examples/ROS/ORB_SLAM2
1
Traceback (most recent call last):
  File "/opt/ros/noetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/noetic/share/ros/core/rosbuild/private.cmake:99 (message):
  [rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is "/home/bzofsv/ORB_SLAM2/Examples/ROS/ORB_SLAM2".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/noetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:5 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/bzofsv/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.

sugnite, 2023-06-22:
This error message implies there's a discrepancy in where ROS is looking for your package, ORB_SLAM2, and where it is actually located. It seems that rospack finds the ORB_SLAM2 package at an empty directory, while the actual package is located in "/home/user/ORB_SLAM2/Examples/ROS/ORB_SLAM2". This suggests that there may be an issue with your ROS_PACKAGE_PATH.

To investigate this issue and find a solution, I'll need some additional information:

1. What is your ROS distribution version? (e.g. Noetic, Melodic, etc.)
2. How did you install ROS and ORB_SLAM2? Did you follow the instruction taking care of every steps ?
3. Please provide the complete output of the following command: `echo $ROS_PACKAGE_PATH`

The output of these commands will help me understand your environment better and provide a suitable solution.

Aishkrish18, 2023-06-22:
Sure. 
1. My ROS Distribution is Noetic running on Ubuntu 20.04
2.  Yes I followed all the instructions as given  in the repository. 
3. ROS_PACKAGE_PATH = /opt/ros/noetic/share   
export ROS_PACKAGE_PATH=/opt/ros/noetic/share:/home/orb slam/ORB_SLAM2/Examples/ROS/ - This is what I added in my bashrc file

sugnite, 2023-06-22:
From what you send it seem that your `$ROS_PACKAGE_PATH` env variable is not containing the full export path that you given in the `~/.bahsrc` as you are missing the path to the package you want to build : `/home/orb slam/ORB_SLAM2/Examples/ROS/`.

Also do you have a ros workspace ? did you build it ?

Aishkrish18, 2023-06-22:
When I perform catkin build inside /home/orb slam/ORB_SLAM2/Examples/ROS/ , it builds the package, after that when I try to rosrun , it says cant find ORB_SLAM2

RAHUL-B-PILLAI, 2023-10-11:
Were any of you able to solve the issue because I am facing similar issues

Aishkrish18, 2023-10-11:
I shifted to docker for orbslam2, and it works better there!

Christmas11235, 2024-09-24:
I just solved this problem, because my default python is 2.7, you can switch to python3 and try it.

supermice, 2024-09-24:
邮件已收到！ ***@***.***

lx-r, 2024-09-24:
您好，来信已收到，谢谢！