raulmur/ORB_SLAM2 #1109: When I run './build.sh', I made an error like ‘recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o' failed’

jas200107, 2024-05-20:
c++: internal compiler error: 已杀死 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
CMakeFiles/ORB_SLAM2.dir/build.make:206: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o] Error 4
make[2]: *** 正在等待未完成的任务....
c++: internal compiler error: 已杀死 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
CMakeFiles/ORB_SLAM2.dir/build.make:494: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o] Error 4
c++: internal compiler error: 已杀死 (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
CMakeFiles/ORB_SLAM2.dir/build.make:86: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 4
CMakeFiles/Makefile2:252: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

sirackerman, 2024-08-12:
did you get any error during the compilling process prior to the error shown at the end?

AndyMRoberts, 2025-01-22:
I have similar errors at the end of my compilation process, did you get this resolved by any chance?